aniruddha-adhikary/mrt-buddy #3: Basic NFC reader feature , No api integrated!

Syed-Bipul-Rahman, 2024-11-03:
### **Basic NFC Reader Feature - No API Integrated**

This individual has only implemented a basic NFC reader feature that reads NFC card information. Unfortunately, they are misleading the users of this app. The data for MRT passes is securely stored on a server. 

He has not integrated any API; instead, he reads the NFC data and displays random values to the user. MRT or Rapid passes are special types of NFC cards that only contain the card number. The actual data, such as the card balance, is stored on a server. To display accurate information from the server, an API must be implemented to fetch the real data.

aniruddha-adhikary, 2024-11-03:
@Syed-Bipul-Rahman Please do study up on Card-Based Accounting systems and FeliCa. Cheers mate.

closed 2024-11-03 by aniruddha-adhikary

Syed-Bipul-Rahman, 2024-11-04:
> @Syed-Bipul-Rahman Please do study up on Card-Based Accounting systems and FeliCa. Cheers mate.

Thank you for your response regarding my issue. I appreciate your guidance and the opportunity to learn more about Card-Based Accounting systems and the FeliCa NFC system.

After conducting further research, **I realize that my understanding of the NFC data and its integration with server-side systems was incorrect.** I apologize for any confusion or frustration my previous comments may have caused.

I value the effort and expertise that you have put into the project, and I look forward to contributing positively to the discussion moving forward.