aniruddha-adhikary/mrt-buddy #4: feat: Add Metro Station Map

kazimmt, 2024-11-03:
You can add a different tab or menu to show the stations map
So people can find the stations easily

Something like this: https://www.google.com/maps/d/u/1/edit?mid=1snBTSj_LQEpBhzfcMTaCD-huo6m63aw

rahat14, 2024-11-04:
@aniruddha-adhikary  need to change our layout, bottom nav, or fragment approach!

chayanforyou, 2024-11-26:
@aniruddha-adhikary dada, I have an idea for it. In this case, we can use open street map (OSM), which will work on both platforms. It is completely free and doesn't require any API keys to use.

I can contribute here if anyone is not working on this.

loius0044, 2025-03-03:
A metro station map is an essential tool for commuters, providing a clear and organized layout of train routes, station locations, and transfer points [Dubai Metro Map](https://dubaimetromap.ae/). It helps passengers navigate complex metro systems efficiently, ensuring they reach their destinations with minimal confusion. Modern metro maps often use color-coded lines and symbols to indicate different routes and services, making them user-friendly for both locals and tourists.

kazimmt, 2025-03-03:
and [this](https://commons.wikimedia.org/wiki/File:Dhaka_MRT_Network_(en).svg).

closed 2025-09-03 by linear[bot]