aniruddha-adhikary/mrt-buddy #12: Updating the app within the app

alaminsoyaib, 2024-11-04:
Add an update app popup so existing users who have installed the app will know there is a new release available and can download it within the app.

aniruddha-adhikary, 2024-11-04:
Perhaps we should go Play Store?

alaminsoyaib, 2024-11-04:
Great 🎉

kazimmt, 2024-11-04:
Send an "Update available! Please update your app" notification at least

aniruddha-adhikary, 2024-11-04:
@kazimmt That's a good idea.

I suppose what worries me is user trust. Like currently there is no internet access for the app. A good assurity that data isn't being sold or collected.

[image]

I am adding the WhatsApp Channel link to announce updates... But what are your thoughts?

rahat14, 2024-11-04:
@aniruddha-adhikary  we can check for the release via GitHub API and can hyper link the user to the git hub page from the app  ?

aniruddha-adhikary, 2024-11-17:
We are on F Droid, Play Store Beta and App Store. So marking this as redundant. Thank you!

closed 2024-11-17 by aniruddha-adhikary