aniruddha-adhikary/mrt-buddy #25: Card Alias

mdminhazulhaque, 2024-11-04:
Hey thanks for this amazing app. I was wondering if you could map the cards serial number with an (human id like npm or docker) alias. Say I have 3 cards with 3 unique serial number. It would be nice if the app could store these cards into an inventory.

For example,

01:22:33:44:55:66:00:01 = Wife
01:22:33:44:55:66:00:07 = Mom
01:22:33:44:55:66:00:08= Kid

 So next time ant card is tapped onto the NFC tag, the app will automatically bring the correct card (with alias) to top, show the latest balance and may be in near future send some popup notification as well if the balance is critical.

Thanks again for this amazing tool.

aniruddha-adhikary, 2024-11-04:
Is the intention for now to identify which one belongs to the mom and which one belongs to the kid? :D

bmabir17, 2024-11-04:
I think , users should be given the choice of at least saving the card data on device. This would in turn help them keep a history of transactions more than 10 that is allowed in the card.

EngrShahria, 2024-11-05:
I think it's unnecessary to do, or have in the app.

The main intention of the MRT-Buddy is to read the NFC memory and provide you with the proper information like balance and last 10 transactions. Tap the card to your phone and get the info.

This feature is useless!

aniruddha-adhikary, 2024-11-05:
I think this is an interesting feature request.

This does extend the apps capability to what I initially thought would be the scope of the app, and it makes it more usable for a group of people.

I would consider this a medium to low priority request for the time being. Once we go through the more urgent tasks like the iOS rewrite, I believe we can look at it with a fresh pair of eyes. 

This is a great suggestion nevertheless thank you everyone!

SamsadSajid, 2024-11-05:
# One possible approach:

## Feature alias

An in-memory cache to store the map of card number to alias.

```go
type struct Alias {
    cardSN string
    alias string
}
```

### Alias(Cache) update:: path 1:
1. When a card is tapped if the card does not exist, show a pop-up to the user to get an alias name. The default value of the alias will be an empty string. So if the user does not provide anything/close the modal, no alias will be shown for the card.
2. Update the cache with the  `cardSN -> alias` entry

### Alias(Cache) update:: path 2:
A different page/tab whichever is smooth from UX pov, showing a UI to the user so that they can **only** update the alias name. The `cardSN` field should be read only in this view. 

If they update the alias name then update the cache with the  `cardSN -> alias` entry

### Noop
While tapping the card if it already exists then no need to update the cache. Therefore, user journey path 1 from the above is not required.

### Data structure size
Assumption made:

- The number of chars for the card serial number: 23 (from the Issue description)
- the max size of the alias is 32 characters
So we need `23+32 = 55 bytes` of memory for one card.

### Security
I don't have any context on whether the card serial number qualifies as PII data. If so, we can encrypt it and use the encrypted value as the key of the cache.

## Feature show the latest tapped card in the top view
> So next time ant card is tapped onto the NFC tag, the app will automatically bring the correct card (with alias) to top

I haven't read the codebase so can't really comment if the app stores the card data in memory. Assuming it shows everything in real-time it does not store any data from the card. In this scenario, we just need to make one extra operation to load the alias name from the cache to the card view. So the flow will be:

card tapped -> fetch alias name with the tapped card's serial number (encrypt the number to form the key if it's the requirement) -> display the alias

How does this sound?

aniruddha-adhikary, 2024-11-15:
Good stuff! Since we had Room already, we went with Room to save the cards!

closed 2024-11-15 by aniruddha-adhikary

mdminhazulhaque, 2025-03-13:
This feature is a reality. Thanks everyone who helped to implement this feature.

[image]