aniruddha-adhikary/mrt-buddy #36: Credits for Contributors

aniruddha-adhikary, 2024-11-05:
MRT Buddy is no longer made by Ani only, so many amazing new contributors joined us! We need to credit them all.

I'm thinking of a simple one.

The list of contributors should be on our GitHub pages website. (I.e, gh-pages branch of this repo). And the app could just launch the browser when users click contributors button or footer.

hmasum52, 2024-11-05:
You can add list of the contributors at the bottom of the README as well! 
Here is how you can do this https://github.com/all-contributors/all-contributors

aniruddha-adhikary, 2024-11-05:
Yeet we have a contributors page and the android app now points to the contributors page!

hmasum52, 2024-11-06:
@aniruddha-adhikary are you doing that manually? I think currently the contributors page misses some contributors. You can automate it using [allcontributors](https://allcontributors.org/docs/en/bot/overview) bot!

aniruddha-adhikary, 2024-11-06:
Ah I wrote a Python script lol

closed 2024-11-06 by aniruddha-adhikary