aniruddha-adhikary/mrt-buddy #37: Flutter Rewrite

aniruddha-adhikary, 2024-11-05:
In order to support the iOS users, we want to maintain a single code base so the experience is coherent across platforms. So flutter could be the answer to create a cross platform experience from a single code base. 

Since the application needs to be written in flutter, we need to be careful on what changes we land on the code base now before the switch happens. 

Any other concerns please do comment.

mdabdulazizzisan, 2024-11-05:
Why not KMM? As the codebase is already in Kotlin, we can try KMM can't we?

hm-tamim, 2024-11-05:
I would also prefer KMP with Compose Multiplatform. And yes, platform-specific code will be required for NFC.

rahat14, 2024-11-05:
Extending @hm-tamim opninion   KMP with Compose Multiplatform will be a good decision as we have android ready ios wont take time  to implement. @aniruddha-adhikary  I am also willing to help on it as I have some experience in it.

aniruddha-adhikary, 2024-11-05:
@rahat14 @hm-tamim @mdabdulazizzisan Should we go for [Compose Multiplatform](https://www.jetbrains.com/compose-multiplatform/) or keep Native UI for each platform?

rahat14, 2024-11-05:
@rahat14  from my perspective, we can share the UI via compose I guess it will be the best decision as the UI will not require anything complex. if in the future there is a situation where things get tight we can always write swift UI and then interpolate with compose via any view class 
.

rahat14, 2024-11-05:
> I am worried how NFC handling will work in KMP. We need to write some duplicate platform specific code?

@aniruddha-adhikary  i was also thinking that , I  will do some study on that tomorrow will come back to this thread inshallah

aniruddha-adhikary, 2024-11-05:
Meanwhile, I will refactor to split out as much platform independent code as possible for you all to start.

rahat14, 2024-11-05:
> @rahat14 Meanwhile, I will refactor to split out as much platform independent code as possible for you all to start.

@aniruddha-adhikary  sure vaia .

aniruddha-adhikary, 2024-11-06:
Okay lol so I started a new KMP project, and ported the existing code inside. Progress is, the android version works. The NFC Logic for iOS um.... not so much. Will create a branch and update here.

aniruddha-adhikary, 2024-11-06:
https://github.com/aniruddha-adhikary/mrt-buddy/tree/refactor/kotlin-multiplatform

Now this does work on Android... but I have zero idea on iOS stuff so would appreciate some help haha.

aniruddha-adhikary, 2024-11-06:
https://github.com/aniruddha-adhikary/mrt-buddy/tree/refactor/kotlin-multiplatform

Now, I do get the balance but... the State of the UI is not updating when I am scanning from iOS. Android appears to work just fine. Some concurrency thread issue I am not understanding. Would appreciate any and all help!

rahat14, 2024-11-06:
@aniruddha-adhikary  vai u r too fast was working on this from last night . will check it now

arafatamim, 2024-11-06:
Speaking of Flutter, I've been working on a similar app that fetches the card info from the website. But I like the approach you've gone with.
I've cleaned up the code and published the repo if you want to take a look. PS, it won't work now because the website is down, so here's a screenshot
https://github.com/arafatamim/rapid_pass_app
[image]

closed 2024-11-15 by aniruddha-adhikary