aniruddha-adhikary/mrt-buddy #39: CI on Pull Requests

aniruddha-adhikary, 2024-11-05:
Can we get the current GitHub Actions to be able to run on a Pull Request (when a maintainer with permissions manually runs the workflow)?

And also like comment out an artifact? It's so that we can download and test the builds.

## Considerations

- Should the signing key be different than the normal signing key use to prevent malicious actors into tricking us into getting malicious code built with our keys?

hmasum52, 2024-11-05:
I think the CI should run the test suits only in separate workflow as github action will be trigger for each push in a PR.

aniruddha-adhikary, 2024-11-05:
Not on each push, but like once a maintainer manually triggers on a PR when they're satisfied @hmasum52

hmasum52, 2024-11-05:
@aniruddha-adhikary  So you are suggesting two workflows
- First one will run the test suits (CI) for a new PR.
- Second one will be triggered manually by a maintainer to build an artifacts to test if everything is working fine!
Correct me if I am wrong!

aniruddha-adhikary, 2024-11-05:
Yes! This is more of a wishlist to do later.

hmasum52, 2024-11-05:
@aniruddha-adhikary As you are planning to do the further development in a cross platform framework, after you finalize, I will be happy to create a PR later!

About framework change. How are you planning to do that? Will you be creating new repository?

closed 2025-05-18 by linear[bot]