aniruddha-adhikary/mrt-buddy #40: The website is kinda basic

aniruddha-adhikary, 2024-11-05:
Anyone has any good design ideas about the website? 

https://mrtbuddy.com

`gh-pages` branch of this repo.

Nafisarkar, 2024-11-05:
https://www.cursor.com what do you think about something similar to this? 🙄

mihf05, 2024-11-05:
https://legendary-marzipan-68df0e.netlify.app/
@aniruddha-adhikary   is it ok

aniruddha-adhikary, 2024-11-05:
Hell yeah this is DOPE. 

Let's remove the testimonial and perhaps use the real screenshot from the correct app!

This looks really clean.

Could we also do a similar theme for the contributors page

enamulkhanbd, 2024-11-06:
I'm a UX/UI Designer, and I’d love to contribute as well. @aniruddha-adhikary aniruddha-adhikary, if you could share some thoughts, it would be great and help me design even better.

closed 2024-11-11 by aniruddha-adhikary