aniruddha-adhikary/mrt-buddy #42: Flutter app build is completed.

mihf05, 2024-11-05:
@aniruddha-adhikary 
Flutter app creation is complete.  Should I pull request it now?  Before this check the flutter app from this link - https://t.me/metroapptest/14

aniruddha-adhikary, 2024-11-05:
Speedy there whoa!

Holy wow

I seem to be having a little issue 

[image]

aniruddha-adhikary, 2024-11-05:
@mihf05 hey man, one more thing. So typically if someone wants to work on a task, there is a process called assigning so essentially you volunteer perhaps and the maintainer says OK. This is the person who is doing it. No one else do it right? Otherwise what could happen is multiple people could be working on the same thing so typically we sound out and once we get the all clear to go ahead, we go ahead.

This is just for the purpose of avoiding conflicts later down the road when we discovered two people are working on the same thing.

For now, we are currently still having a discussion in another issue to find out whether we should use Kotlin multi platform or flutter. 😓 😅 

Your pull request is a bit too early whoops

mihf05, 2024-11-05:
@aniruddha-adhikary  Thanks for your discussion and planning. Whichever technology is chosen, I am confident that we will reach a good conclusion. I look forward to updates and am ready to help in any way I can. Love to see the progress of our team's work!

mihf05, 2024-11-05:
maybe bug fixed https://t.me/metroapptest/15

aniruddha-adhikary, 2024-11-06:
@mihf05 Since the folks chose Kotlin, I'm closing this for now. Good work!

closed 2024-11-06 by aniruddha-adhikary