aniruddha-adhikary/mrt-buddy #57: App Crash

aniruddha-adhikary, 2024-11-07:
```
Exception java.lang.RuntimeException: Unable to destroy activity {net.adhikary.mrtbuddy/net.adhikary.mrtbuddy.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'android.view.Window android.app.Activity.getWindow()' on a null object reference
  at android.app.ActivityThread.performDestroyActivity (ActivityThread.java:5109)
  at android.app.ActivityThread.handleDestroyActivity (ActivityThread.java:5139)
  at android.app.servertransaction.DestroyActivityItem.execute (DestroyActivityItem.java:44)
  at android.app.servertransaction.TransactionExecutor.executeLifecycleState (TransactionExecutor.java:176)
  at android.app.servertransaction.TransactionExecutor.execute (TransactionExecutor.java:97)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:2066)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:223)
  at android.app.ActivityThread.main (ActivityThread.java:7664)
  at java.lang.reflect.Method.invoke
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:592)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:947)
Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'android.view.Window android.app.Activity.getWindow()' on a null object reference
  at android.nfc.NfcActivityManager$NfcActivityState.<init> (NfcActivityManager.java:125)
  at android.nfc.NfcActivityManager.getActivityState (NfcActivityManager.java:176)
  at android.nfc.NfcActivityManager.disableReaderMode (NfcActivityManager.java:226)
  at android.nfc.NfcAdapter.disableReaderMode (NfcAdapter.java:1603)
  at net.adhikary.mrtbuddy.nfc.NFCManager.stopScan (NfcManager.android.kt:122)
  at net.adhikary.mrtbuddy.nfc.NFCManager$startScan$lambda$3$lambda$2$$inlined$onDispose$1.dispose (Effects.kt:491)
  at androidx.compose.runtime.DisposableEffectImpl.onForgotten (Effects.kt:87)
  at androidx.compose.runtime.CompositionImpl$RememberEventDispatcher.dispatchRememberObservers (Composition.kt:1345)
  at androidx.compose.runtime.CompositionImpl.dispose (Composition.kt:787)
  at androidx.compose.ui.platform.WrappedComposition.dispose (Wrapper.android.kt:169)
  at androidx.compose.ui.platform.WrappedComposition.onStateChanged (Wrapper.android.kt:177)
  at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent (LifecycleRegistry.jvm.kt:320)
  at androidx.lifecycle.LifecycleRegistry.backwardPass (LifecycleRegistry.jvm.kt:272)
  at androidx.lifecycle.LifecycleRegistry.sync (LifecycleRegistry.jvm.kt:289)
  at androidx.lifecycle.LifecycleRegistry.moveToState (LifecycleRegistry.jvm.kt:142)
  at androidx.lifecycle.LifecycleRegistry.handleLifecycleEvent (LifecycleRegistry.jvm.kt:124)
  at androidx.lifecycle.ReportFragment$Companion.dispatch$lifecycle_runtime_release (ReportFragment.android.kt:190)
  at androidx.lifecycle.ReportFragment$LifecycleCallbacks.onActivityPreDestroyed (ReportFragment.android.kt:146)
  at android.app.Activity.dispatchActivityPreDestroyed (Activity.java:1498)
  at android.app.Activity.performDestroy (Activity.java:8235)
  at android.app.Instrumentation.callActivityOnDestroy (Instrumentation.java:1344)
  at androidx.test.runner.MonitoringInstrumentation.callActivityOnDestroy (MonitoringInstrumentation.java:1)
  at android.app.ActivityThread.performDestroyActivity (ActivityThread.java:5094)
```

rahat14, 2024-11-08:
@aniruddha-adhikary  could you  tell me  how to reproduce this

aniruddha-adhikary, 2024-11-15:
Investigated and fixed!

closed 2024-11-15 by aniruddha-adhikary