aniruddha-adhikary/mrt-buddy #58: The app is no longer automatically launched upon pressing the card on the back of the Phone.

refat-ul-islam, 2024-11-07:
I was using version 0.6 till today. Whenever I used to press the Rapid pass in the back of my phone it automatically opened the MRT Buddy  app but after updating to the version 0.8 today,  it no longer automatically open the app upon pressing the card. I have to go to the app every time, then it works fine. 
Can you please fix the issue? 
[image]

Zamiul-rashid, 2024-11-07:
I think this was purposefully disabled as it was opening due to detecting other cards with NFC tags as well.

Zamiul-rashid, 2024-11-07:
as the saying goes its not a bug its a feature , haha but let's see what the moderators have to say on this.

refat-ul-islam, 2024-11-07:
> I think this was purposefully disabled as it was opening due to detecting other cards with NFC tags as well.

If you intend to use other cards with NFC, I guess you'll have individual apps for them as well and if you have multiple apps, the phone will ask which one to open.

N:B: As far as I know there isn't much chance of opening an app if the card isn't supported by the app. So,  that individual card have to supported by the MRT Buddy app first to occur accidental opening.

aniruddha-adhikary, 2024-11-17:
We have chosen to block this for now. People who keep their wallet and phone in the same pocket had erratic launches.

closed 2024-11-17 by aniruddha-adhikary