aniruddha-adhikary/mrt-buddy #60: Add metro timings

GlyX0c0den, 2024-11-07:
Add metro timings in the app

Zamiul-rashid, 2024-11-07:
@aniruddha-adhikary This is a good issue this should get some priority.

shahidx0x, 2024-11-08:
yep, that's what I am thinking about. when this apps will get enough exposure, DMTCL might reach out and provide the developer necessary information to build this feature, that will be great feature for this app

alrafiabdullah, 2024-11-08:
BTW you can see the timings on Google Maps.

GlyX0c0den, 2024-11-08:
I know I can see timings in google maps but,I think as you are making an
app if you add timings in the app it will increase the credibility of the
app.It will be more useful for the users also as they will get all the
services they need at one place.

Zamiul-rashid, 2024-11-08:
As a regular user of Dhaka metro I saw that the timing change based on crowd and the pressure on the trains it isn't consistent say every 5 minutes a train leaves, it's sometimes 5 sometimes 7 sometimes 10 ,so we can't code that without the help of officials ,as it is a community project we don't have access to official timings and stuff.. We have this feature in mind but I would say this is a mid priority feature as out app philosophy for now is to keep it offline (no internet connection) app... I hope you understand..

Thanks for this much support and positive ideas. We really appreciate it..

GlyX0c0den, 2024-11-08:
I understand, thanks for the clarification.Wish you and the team all the
best for the project.🤎🤎

aniruddha-adhikary, 2024-11-11:
Due to the unavailability of reliable data... we have to skip this for now. Closing until this comes up again.

closed 2024-11-11 by aniruddha-adhikary