aniruddha-adhikary/mrt-buddy #62: Dark mode for Website

Zamiul-rashid, 2024-11-07:
We can work on adding darkmode for the website that would bring some quality of life improvements.

anikbsmrstucse, 2024-11-08:
i was reviewing your gh-pages-news branch, and that branch already start landing page development with next.js technology. is there any chance to contribute to the running website, implementing dark mode feature?

Zamiul-rashid, 2024-11-08:
I think someones already working on it.

anikbsmrstucse, 2024-11-08:
if you give me permission I will do it.

anikbsmrstucse, 2024-11-08:
https://jmp.sh/s/MzzCyHFmOG4REHJzFkaT
 here is the dark mode glimps.is it ok?

aniruddha-adhikary, 2024-11-09:
@anikbsmrstucse YO MAN send a PR

anikbsmrstucse, 2024-11-09:
Sir I am doing PR request for dark mode
check it,when you are avaliable

Eniamza, 2024-11-09:
This looks extremely good.

anikbsmrstucse, 2024-11-09:
Thanks a lot sir❤

closed 2024-11-09 by aniruddha-adhikary