aniruddha-adhikary/mrt-buddy #64: Bug: Transaction History doesn't shows 10th Transaction Amount

kazimmt, 2024-11-08:
On Transaction history page, it shows 10 history but shows fare only on 9 transaction.
It doesn't shows fare on 10th one.
[image]

EngrShahria, 2024-11-08:
I can confirm this bug.

kazimmt, 2024-11-11:
Look like it fixed on latest versions 0.0.11 (Showing 19 Transactions)
Can you confirm it @EngrShahria

kazimmt, 2024-11-21:
App version: 0.0.21
[image]

aniruddha-adhikary, 2024-11-21:
It is not possible to show the last amount because...

On the card itself it records the balance. We subtract the nth transaction from the n+1th transaction to get an amount. So since there is no n+1 on the last item, it is N/A.

aniruddha-adhikary, 2024-11-21:
You could choose to hide the line entirely. Or just show whatever information you have haha

kazimmt, 2024-11-21:
I think it's better to hide the entire line

demo-Ashif, 2024-12-25:
Any update on this issue? Any decision taken to hide the line? @kazimmt

kazimmt, 2024-12-25:
Not yet

MeSagor, 2026-10-10:
As the issue is open yet, can you review this PR(#149).