aniruddha-adhikary/mrt-buddy #77: fare calculator

mrjahid11, 2024-11-10:
there is a mistake. almost from every station next to three stations fare is 20 taka. 
for example- Karwan Bazar to Agargaon’s fare 20 taka. but it shows 30 taka. and from farmgate to Agargaon’s fare 20 taka, shows 30 taka. 
[image]

aniruddha-adhikary, 2024-11-10:
Thanks for letting us know. 

The iOS update has been pushed to Apple waiting for approval. Within 24 hours it should reach your device. 

I'll update you here once the update is available.

aniruddha-adhikary, 2024-11-11:
Update pushed as v0.0.10. Apologies for the inconvenience caused and thanks for reporting @mrjahid11!