aniruddha-adhikary/mrt-buddy #80: Requesting to add rules of Metro.

mustarikulislam, 2024-11-11:
I remember checking in and out of the same station only to be slapped a BDT 100 taka charge as the 5 minutes limit to check out was over once I enter the station. Many people unknowingly face such unknown charges. I was hoping the app would add an abridged set of rules that might help many users to avoid such penalties and also the do's and dont's.

BTW, I am really enjoying the app. Best wishes!

aniruddha-adhikary, 2024-11-11:
This is a good suggestion. 

However we do need the product people to figure out how this could possibly be executed. 

I'm thinking something beyond just listing the rules something more useful.

mustarikulislam, 2024-11-11:
Glad that you're giving a thought on this!

closed 2025-05-18 by linear[bot]