aniruddha-adhikary/mrt-buddy #81: feat: Provide a way to open NFC settings when device has NFC

Mahmud0808, 2024-11-11:
**Description:**
Implement a feature that automatically directs users to the NFC settings page when the device supports NFC.

**Visual Representation:**
<div align="center">
    [image]
</div>

**Proposed Implementation:**

  - The app should detect if the device supports NFC.
  - If NFC is supported, the app should provide a way for the user to open the NFC settings directly.
  - The following Kotlin function can be used to open the NFC settings on Android:

```kotlin
private fun openNfcSettings() {
    val intent = if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.Q) {
        Intent(Settings.Panel.ACTION_NFC)
    } else {
        Intent(Settings.ACTION_WIRELESS_SETTINGS)
    }
    startActivity(intent)
}
```

**Benefits:**

- Simplifies the process of enabling or configuring NFC for users.

aniruddha-adhikary, 2024-11-15:
This is a great feature request, we should prioritize.

Zamiul-rashid, 2024-11-21:
@aniruddha-adhikary I will start on this if its not fixed already

aniruddha-adhikary, 2024-11-21:
@Zamiul-rashid please go ahead

demo-Ashif, 2024-12-25:
Any update on this @Zamiul-rashid ?

closed 2025-06-25 by linear[bot]