aniruddha-adhikary/mrt-buddy #86: bug: App needs restart to scan a card

Mahmud0808, 2024-11-12:
**Issue Description:**

When the app is opened with NFC disabled on the phone, it correctly indicates that NFC is disabled. However, after enabling NFC from the settings (keeping the app running in the background), attempting to scan a card results in the device vibrating but no UI changes are observed. The UI only displays "Tap Your Card." Repeated card tapping causes vibration each time but no further UI updates. Closing and reopening the app resolves the issue, allowing successful card scanning. This problem does not occur if NFC is enabled before opening the app.

**Steps to Reproduce:**

1. Open the app with NFC disabled on the phone.
2. Enable NFC from the phone settings while keeping the app running in the background.
3. Attempt to scan a card.
4. Observe that the device vibrates but no UI changes occur.

**Expected Behavior:**

The app should recognize the card and update the UI with the card information. If a restart is necessary, the app should prompt the user to restart the app.

**Actual Behavior:**

- Device vibrates when the card is tapped.
- No changes occur on the UI except the "Tap Your Card" message.
- Closing and reopening the app resolves the issue, and card information is displayed after tapping.

**Error Log:**

```
2024-11-12 19:59:02.504  1570-2614  ActivityTaskManager     system_server                        E  With Android 15 BAL hardening this activity start may be blocked if the PI creator upgrades target_sdk to 35+!  (missing opt in by PI creator)![callingPackage: net.adhikary.mrtbuddy; callingPackageTargetSdk: 34; callingUid: 10784; callingPid: -1; appSwitchState: 2; callingUidHasAnyVisibleWindow: true; callingUidProcState: TOP; isCallingUidPersistentSystemProcess: false; forcedBalByPiSender: BSP.NONE; intent: Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x20000000 cmp=net.adhikary.mrtbuddy/.MainActivity (has extras) }; callerApp: null; balAllowedByPiCreator: BSP.ALLOW_BAL; balAllowedByPiCreatorWithHardening: BSP.ALLOW_BAL; resultIfPiCreatorAllowsBal: BAL_ALLOW_VISIBLE_WINDOW; hasRealCaller: true; isCallForResult: false; isPendingIntent: true; autoOptInReason: null; realCallingPackage: com.android.nfc; realCallingPackageTargetSdk: 34; realCallingUid: 1027; realCallingPid: 3931; realCallingUidHasAnyVisibleWindow: false; realCallingUidProcState: PERSISTENT; isRealCallingUidPersistentSystemProcess: true; originatingPendingIntent: PendingIntentRecord{805b5a net.adhikary.mrtbuddy startActivity}; realCallerApp: null; balAllowedByPiSender: BSP.ALLOW_FGS; resultIfPiSenderAllowsBal: BAL_BLOCK]
```

**Device Information:**

- ROM: AOSP Android 14
- Security Patch Level: October 2024

aniruddha-adhikary, 2024-11-12:
I could reproduce the bug on Android 14.

Filing as actual bug.

aniruddha-adhikary, 2024-11-15:
This bug has been fixed in the latest release. Thanks @Mahmud0808 !

closed 2024-11-15 by aniruddha-adhikary