aniruddha-adhikary/mrt-buddy #88: Missing minute time in the class "TimestampService"

EngrShahria, 2024-11-13:
The timestamp of each transaction missing the minute.

I have checked the TimeStampService Class, it has no minute function.

`kotlin

    fun decodeTimestamp(value: Int): LocalDateTime {
        val hour = (value shr 3) and 0x1F
        val day = (value shr 8) and 0x1F
        val month = (value shr 13) and 0x0F
        val year = (value shr 17) and 0x1F

        // Calculate the actual year
        val fullYear = 2000 + year

        // Validate month and day
        val validMonth = if (month in 1..12) month else 1
        val validDay = if (day in 1..31) day else 1

        // Create a LocalDateTime instance
        return LocalDateTime(
            year = fullYear,
            monthNumber = validMonth,
            dayOfMonth = validDay,
            hour = hour % 24,
            minute = 0, // RETURNS 0 ALWAYS, NO function to get the minute time.
            second = 0,
            nanosecond = 0
        )
    }
`

[image]

Please fix it.

aniruddha-adhikary, 2024-11-13:
Hi @EngrShahria we couldn't find the offset for finding the minute value. The card holds data to the hour, from what we investigated thus far.

Do let us know if you found anything.

aniruddha-adhikary, 2024-11-15:
Closing due to inactivity.

closed 2024-11-15 by aniruddha-adhikary

Mahmud0808, 2024-11-15:
My suggestion is to add the word 'around' before the time so that it doesn't confuse other users.